scsitape/stenc #11: stenc + CentOS 7.3, 7.4 and 7.5: Sense Code: Illegal Request (0x05) on LTO drive (encryption)

devnullzero, 2018-08-26:
Under CentOS 7.3 to 7.5 `stenc` returns the following error:

> $ stenc -f /dev/nst0 -a 1 -e on -k test.key
> Provided key length is 256 bits.
> Key checksum is 7a43.
> Turning on encryption on device '/dev/nst0'...
> Sense Code:              Illegal Request (0x05)
>  ASC:                    0x26
>  ASCQ:                   0x00
>  Additional data:        0x00000000000000000000000000000000
> Error: Turning encryption on for '/dev/nst0' failed!
> Usage: stenc --version | -g <length> -k <file> [-kd <description>] | -f <device> [--detail] [-e <on/mixed/rawread/off> [-k <file>] [-kd <description>] [-a <index>] [--protect | --unprotect] [--ckod] ]
> Type 'man stenc' for more information.

Adding the line: `memset(&kad,0,sizeof(kad));` in `scsiencrypt.cp` fix this issue as described here:

What could cause a 'sense error' when setting LTO encryption?
https://serverfault.com/questions/864580/what-could-cause-a-sense-error-when-setting-lto-encryption

> SCSIWriteEncryptOptions():
> SSP_KAD kad;
> memset(&kad,0,sizeof(kad));   <--------
> kad.type=0x00;
> 

Thanks to lan and Kasperd for the solution.

remcohn, 2018-09-06:
solves the issue on Debian8 with Quantium LTO6 drive too. thanks a lot!

bahelit, 2018-10-23:
Attempting to enable encryption on an IBM tape library 3573-TL from a machine running CenOS 7.4 and receive an identical error even after applying the patch.

`# stenc -f /dev/nst0 -a 1 -e on -k /etc/tape.key 
Provided key length is 256 bits.
Key checksum is 4637.
Turning on encryption on device '/dev/nst0'...
Sense Code:              Illegal Request (0x05)
 ASC:                    0x24
 ASCQ:                   0x00
 Additional data:        0x00000001534c4f5430303000000007c7e1000000000080b560000000000000000000000000000000000000000000000000000000000000000000000000000000000000003039355133303439363600000000000000000000
Error: Turning encryption on for '/dev/nst0' failed!
`

jonasstein, 2019-04-16:
@ninthclowd what do you think, should we merge this?

scsirob, 2019-09-17:
@bahelit Your error is different. ASC 0x26 means your drive rejects one of the SCSI commands it got from stenc. The original error referred to ASC 0x24 which means the SCSI command was accepted but the data sent by that command wasn't.

kedros-hrubsa, 2019-09-19:
We also need the patch on Debian 10 amd64.
Our tape drive is Quantum Ultrium LTO-5.

Please merge the patch. I think there can be nothing wrong with zeroing the memory before using it...

scsirob, 2020-05-28:
When you zero the entire kad structure, the two lines below it to set kad.type and kad.authenticated to zero can be removed.

royalmoose, 2020-06-06:
> Adding the line: `memset(&kad,0,sizeof(kad));` in `scsiencrypt.cpp` fix this issue as described here:
> 
> > What could cause a 'sense error' when setting LTO encryption?
> > https://serverfault.com/questions/864580/what-could-cause-a-sense-error-when-setting-lto-encryption
> 
> Patch for stenc v1.0.7 (https://github.com/scsitape/stenc/archive/1.0.7.tar.gz):

Just tested on Debian 9 and 10 with Quantum LTO5. Patch works perfectly.

PR #17 (2020-11-14): Remove duplicate zeroing
memset zeroes whole &kad. No need to set it's parts to zero.

Correctly pointed out by @scsirob at https://github.com/scsitape/stenc/issues/11#issuecomment-635454013.

The patch from #11 is already merged in master. This PR is an upgrade to that and it should close #11.
closed 2020-11-14 by ninthclowd