scsitape/stenc #15: Unable to activate mixed mode ecryption

marianoemon, 2019-02-12:
Hi all, 
I'l trying to implement stenc for bacula, in particular on LTO6 unit and seems this option is not compatible. Could be this true?
Ubuntu Server 18.04.2, stenc 1.0.7-2

stenc -f /dev/nst0 -e mixed -k /etc/bacula/                                                                                                                                                             myaes.key -a 1 --ckod
Provided key length is 256 bits.
Key checksum is ffffcad2.
Turning on encryption on device '/dev/nst0'...
Sense Code:              Illegal Request (0x05)
 ASC:                    0x26
 ASCQ:                   0x00
 Additional data:        0x00000e00202020202020201500005881b9000097f0b982b5504c3                                                                                                                                                             60000000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                             00045573233584d3250314100000000000000000000
Error: Turning encryption on for '/dev/nst0' failed!
Usage: stenc --version | -g <length> -k <file> [-kd <description>] | -f <device>                                                                                                                                                              [--detail] [-e <on/mixed/rawread/off> [-k <file>] [-kd <description>] [-a <inde                                                                                                                                                             x>] [--protect | --unprotect] [--ckod] ]
Type 'man stenc' for more information.
root@mxmexbkp01:/etc/bacula/scripts# Error: Turning encryption on for '/dev/nst0                                                                                                                                                             ' failed!
Error:: command not found

jonasstein, 2019-12-25:
I use stenc together with an Tandberg LTO-6 drive. 
I was told that there are very few LTO-6 drives, which do not support AES just to add more products on the marked. But I could not confirm this yet. Which exact drive do you use?
Can you explain how bacula adds the AES support now? Does it really use the AES chip in the LTO drive?

jonasstein, 2020-06-16:
@marianoemon ping, any news?

svalx, 2020-12-01:
Probably incorrect encryption algorithm for drive. Try without -a option or another value. I receive such error on HP drive when set -a option other than one.