scsitape/stenc #24: Can we get AIX binary for 1.0.8?

c3rberus, 2020-09-29:
There is a new 1.0.8 release but no aix7_binary_stenc-1.0.8. Is it possible to attach a binary version for us that don't have access to an AIX box with builds tools. Would be very useful.

jonasstein, 2020-11-09:
I have no AIX, but I think @ninthclowd has access to AIX. We need volunteers who compile and test binaries.

ninthclowd, 2020-11-10:
unfortunately I am in a different job now and don't have access to an AIX system anymore

c3rberus, 2021-05-11:
I have access to AIX 7.2, if you can you share steps to compile in AIX I can give it a shot. The AIX install is bare bone, I don't think it has any dev tools on there but I can give it a shot.

ninthclowd, 2021-05-11:
@c3rberus, it has been a long time since I originally compiled the binaries on AIX, but I seem to remember installing various  build tools from https://www.ibm.com/support/pages/aix-toolbox-linux-applications-downloads-alpha.   I would try installing the following:
* gcc
* gcc-c++ 
* autoconf
* make (GNU)

From there I believe you can do the following:
autoreconf --install
./configure --with-static-libgcc
gmake

That should get you the binary.

jonasstein, 2022-05-03:
We still have no one with AIX. See also https://github.com/scsitape/stenc/issues/54

jonasstein, 2023-05-01:
Closing here, as we have no AIX available here. Please join the [beta testers](https://github.com/scsitape/stenc/issues/36), if you can help out with AIX.

closed 2023-05-01 by jonasstein