scsitape/stenc #29: MacOS version

albertogarciacz, 2022-03-06:
Hello, is there a chance to run this tool on MacOS?

jonasstein, 2022-03-06:
I have no OSX available. Please report, if you can test it.

albertogarciacz, 2022-03-06:
I have MacOS available. I can possibly test it.

albertogarciacz, 2022-03-06:
I tried to manually compile from source. However, when I ran ./configure the below mentioned error came up. 
`checking your OS... configure: error: unknown OS type: Darwin`
I guess that code has to be adjusted for MacOS.

jmwilson, 2022-05-02:
According to the following, I don't think this is doable:

- no built-in SCSI tape driver
- no `mtio.h` header file for tape control ioctls
- no `mt` command available for doing everything else you'd need

I've seen some commercial utilities that do LTFS for MacOS, and they are maybe supplying their own DriverKit extension to interface with the drive. I think doing the same would be out of scope for stenc... have you tested connectivity of a tape drive using a bare-metal SAS HBA or iSCSI initiator on MacOS?

https://apple.stackexchange.com/questions/380390/is-it-possible-to-install-the-gnu-mt-tape-drive-command-in-osx