scsitape/stenc #32: Get good random numbers by c++ function/lib, not via OS

jonasstein, 2022-04-20:
To not pull random numbers via OS.

Two possible solutions:

**a) Unix philosophy: One tool per task.**
we drop the key generation by `stenc` and just provide a manual with shell commands to generate a secure key and explain how to test the random source.

**b) use a proper C++ library** 
Also include a parameter, which will send a huge amount of the used random to `stdout` for test purpose:

    # user can test, if his library and random source are working.
    $> stenc --test-rnd | entropy_checker
    Shannon entropy is 8.8

jonasstein, 2022-04-24:
(b) seems to cause the most work for us without any benefit. I like option (a) most. Users can use their beloved tool to generate the random key. Either a hardware TRNG, a set of tools on the commandline, or whatever. Our key generator will never be better than what the user already has. 
But this change should go to a major milestone release, if we go this way. @ninthclowd what Do you think about it?

ninthclowd, 2022-04-25:
Personally I like your idea of dropping key generation and leave it to tools that are built for that purpose.  Maybe just update the man pages with an example of how to do so with openssl?

closed 2022-05-09 by jonasstein