scsitape/stenc #35: LGTM CI Checks fail without AC_CHECK_INCLUDES_DEFAULT

jonasstein, 2022-04-21:
LGTM seems to need a fix, if  `AC_CHECK_INCLUDES_DEFAULT` is not active.
The macro was dropped due to https://github.com/scsitape/stenc/issues/33

```
Run autoreconf -i
  autoreconf -i
  shell: /usr/bin/bash -e {0}
configure.ac:18: error: possibly undefined macro: AC_CHECK_INCLUDES_DEFAULT
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
Error: Process completed with exit code 1.
```

sunwire, 2022-04-22:
This has failed because Github CI uses Ubuntu 20.04 which uses autoconf version 2.69. And this macro requires version 2.70
And it has failed **WITH** this macro not **WITHOUT**.
Macro introduced, CI failed for the first time https://github.com/scsitape/stenc/commit/aa2244308dc947bce16b5ad1e4a528a21553bdcc
Macro dropped, all CI test passed https://github.com/scsitape/stenc/commit/0ad0a208d5602b0327cbc7e879a8b665805111ca

jonasstein, 2022-04-23:
Ah yes, seems I did not wait long enough. After removing `AC_CHECK_INCLUDES_DEFAULT` the build failed. 
Then we can close this one.

closed 2022-04-23 by jonasstein