scsitape/stenc #39: Errors don't display correctly

sunwire, 2022-04-24:
When an error appears, version  1.0.8 displays:
```
Sense Code:              Illegal Request (0x05)
 ASC:                    0x26
 ASCQ:                   0x00
 Additional data:        0x00000000000000000000000000000000
```
but version 1.0.9 and 1.1.0 displays:
```
Sense Code:              Illegal Request (0x)
 ASC:                    0x&
 ASCQ:                   0x
 Additional data:        0x
```
The reason for this are changes made in https://github.com/scsitape/stenc/commit/aa2244308dc947bce16b5ad1e4a528a21553bdcc
The old code uses HEX() macro https://github.com/scsitape/stenc/blob/844306dbe4ac29ffacf29d09f68338cfb3b8fd2b/src/scsiencrypt.h#L39-L40 it cast to the appropriate type (int), but a new code doesn't do it. e.g:
old:
`cerr<<" (0x"<<HEX(sd->senseKey)<<")"<<endl;`
new:
`std::cerr<<" (0x"<<std::hex << (sd->senseKey);`
it should be `std::cerr<<" (0x"<<std::hex << int(sd->senseKey);` or `std::cerr<<" (0x"<<std::hex << static_cast<int>(sd->senseKey);`
At least 7 lines of code are affected.

jonasstein, 2022-04-24:
I think we should drop that macro and write a proper function with test.

jonasstein, 2022-04-24:
Seems we are all working on the same part now ;-) Ideally we would chat and coordinate, if you have time now, I will join jitsii on https://jitsi.fem.tu-ilmenau.de/stenc and we can discuss how to proceed @sunwire  @jmwilson

sunwire, 2022-04-24:
@jonasstein Unfortunately, I do not have time now.

jonasstein, 2022-04-24:
ok

jonasstein, 2022-04-24:
fixed by revert in https://github.com/scsitape/stenc/commit/f1a36eba4236fae198f1564eb92b9975d16647af

closed 2022-04-24 by jonasstein