scsitape/stenc #42: Workflow to run catch offline

jonasstein, 2022-04-24:
How should a workflow look like (with the exact commands) to run the catch test framework offline?

jmwilson, 2022-04-24:
With autotools, you can run `make check` from the project root directory.

jonasstein, 2022-04-24:
Do I miss something?

```
make check
Making check in src
make[1]: Entering directory '/home/stein/my/prj/stenc2/stenc/src'
depbase=`echo main.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..      -g -O2 -MT main.o -MD -MP -MF $depbase.Tpo -c -o main.o main.cpp &&\
mv -f $depbase.Tpo $depbase.Po
depbase=`echo scsiencrypt.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..      -g -O2 -MT scsiencrypt.o -MD -MP -MF $depbase.Tpo -c -o scsiencrypt.o scsiencrypt.cpp &&\
mv -f $depbase.Tpo $depbase.Po
depbase=`echo keyinfo.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..      -g -O2 -MT keyinfo.o -MD -MP -MF $depbase.Tpo -c -o keyinfo.o keyinfo.cpp &&\
mv -f $depbase.Tpo $depbase.Po
g++   -g -O2   -o stenc main.o scsiencrypt.o keyinfo.o  
make[1]: Leaving directory '/home/stein/my/prj/stenc2/stenc/src'
Making check in man
make[1]: Entering directory '/home/stein/my/prj/stenc2/stenc/man'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/home/stein/my/prj/stenc2/stenc/man'
Making check in tests
make[1]: Entering directory '/home/stein/my/prj/stenc2/stenc/tests'
make[1]: *** No rule to make target 'check'.  Stop.
make[1]: Leaving directory '/home/stein/my/prj/stenc2/stenc/tests'
make: *** [Makefile:357: check-recursive] Error 1
```

jmwilson, 2022-04-24:
Maybe you need to re-run automake/autoconf stuff? Try `./autogen.sh && ./configure && make check`. This is the output I see on freshly cloned copy:

```
% make check
Making check in src
make[1]: Entering directory '/tmp/stenc/src'
depbase=`echo main.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..      -g -O2 -MT main.o -MD -MP -MF $depbase.Tpo -c -o main.o main.cpp &&\
mv -f $depbase.Tpo $depbase.Po
depbase=`echo scsiencrypt.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..      -g -O2 -MT scsiencrypt.o -MD -MP -MF $depbase.Tpo -c -o scsiencrypt.o scsiencrypt.cpp &&\
mv -f $depbase.Tpo $depbase.Po
depbase=`echo keyinfo.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..      -g -O2 -MT keyinfo.o -MD -MP -MF $depbase.Tpo -c -o keyinfo.o keyinfo.cpp &&\
mv -f $depbase.Tpo $depbase.Po
g++   -g -O2   -o stenc main.o scsiencrypt.o keyinfo.o
make[1]: Leaving directory '/tmp/stenc/src'
Making check in man
make[1]: Entering directory '/tmp/stenc/man'
make[1]: Nothing to be done for 'check'.
make[1]: Leaving directory '/tmp/stenc/man'
Making check in tests
make[1]: Entering directory '/tmp/stenc/tests'
make  scsi
make[2]: Entering directory '/tmp/stenc/tests'
depbase=`echo scsi.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..  -I../src   -g -O2 -MT scsi.o -MD -MP -MF $depbase.Tpo -c -o scsi.o scsi.cpp &&\
mv -f $depbase.Tpo $depbase.Po
depbase=`echo ../src/scsiencrypt.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
g++ -DHAVE_CONFIG_H -I. -I..  -I../src   -g -O2 -MT ../src/scsiencrypt.o -MD -MP -MF $depbase.Tpo -c -o ../src/scsiencrypt.o ../src/scsiencrypt.cpp &&\
mv -f $depbase.Tpo $depbase.Po
g++  -g -O2   -o scsi scsi.o ../src/scsiencrypt.o
make[2]: Leaving directory '/tmp/stenc/tests'
make  check-TESTS
make[2]: Entering directory '/tmp/stenc/tests'
make[3]: Entering directory '/tmp/stenc/tests'
PASS: scsi
============================================================================
Testsuite summary for stenc 1.1.0
============================================================================
# TOTAL: 1
# PASS:  1
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
make[3]: Leaving directory '/tmp/stenc/tests'
make[2]: Leaving directory '/tmp/stenc/tests'
make[1]: Leaving directory '/tmp/stenc/tests'
make[1]: Entering directory '/tmp/stenc'
make[1]: Nothing to be done for 'check-am'.
make[1]: Leaving directory '/tmp/stenc'
```

jonasstein, 2022-04-24:
thank you. Works perfect. Added this to the README

closed 2022-04-24 by jonasstein