scsitape/stenc #43: Consider breaking changes for v2

jmwilson, 2022-04-24:
Follow-up to conversation with @jonasstein. Changing output and logging may be a breaking change, so consider bumping the major version to 2 with additional changes:

- changing logging from /var/log/stenc to standardized syslog (see #31) may be considered breaking
- review proper destination for informational messages and output (stdout vs stderr)
- revise command syntax to be consistent with the `mt(1)` command - I like moving from `stenc -e on` to `stenc encryption on`, and from `stenc --details` to `stenc status`

jonasstein, 2023-05-01:
@jmwilson Do you want to prepare a PR to complete this for 2.0.0?

jmwilson, 2023-05-01:
All of this has been done already, so closing.

closed 2023-05-01 by jmwilson