scsitape/stenc #44: Consider compiling the man page from markdown and friends

jonasstein, 2022-04-24:
The man page is very useful and well written. 
However it is difficult to maintain. We have a PR claiming there would be a typo (I do not think this is true). 

What do you think about a conversion to something like markdown and convert it with `pandoc` as described on https://www.howtogeek.com/682871/how-to-create-a-man-page-on-linux/
This way it is easy to maintain and we get also a nice man/pdf/html/whatever output at once.

Any objections or ideas about the man page, @ninthclowd?

ninthclowd, 2022-04-25:
No, I think that's a great idea!

jonasstein, 2022-04-25:
added https://github.com/scsitape/stenc/blob/master/man/stenc.rst 
to generate man pages with pandoc
Usage:
pandoc --from rst --to man --file stenc.rst -o stenc.1

Could anyone add this to the makefiles?

closed 2022-04-30 by jonasstein