scsitape/stenc #49: Considering C++17 as minimum requirement

jonasstein, 2022-04-29:
We could make the code a lot cleaner with features from C++ 17 or even C++ 20
Which version should we expect?

Seems that even the most important C++ 20 functions have been available since many years in the major compilers:
https://en.cppreference.com/w/cpp/compiler_support

As long we do not use exotic features, it should even be fine to switch to 20. What do you think?
We could also go for 17 now and wait a few years with 20...

jmwilson, 2022-04-29:
I tried C++20 without problems, but consider https://gcc.gnu.org/projects/cxx-status.html (C++20 still is still officially "experimental") and https://clang.llvm.org/cxx_status.html (C++ still with partial support).

My main concern is that getting the best language support means compiling with very recent compiler builds like GCC 11 or 12 and many distros do not follow the bleeding edge.

jonasstein, 2022-04-29:
I read this as a vote for
[x] C++ 17
correct, @jmwilson?

jmwilson, 2022-04-29:
Yep!

jonasstein, 2022-05-03:
@ninthclowd any objections to C++17 as minimum?

jonasstein, 2022-05-08:
No objections claimed so far. Lets go with C++ 17.

closed 2022-05-08 by jonasstein