scsitape/stenc #52: CI needs pandoc

jonasstein, 2022-04-30:
Any ideas how to fix our CI?

Perhaps this can help us:
https://github.com/pandoc/dockerfiles#github-actions

sunwire, 2022-04-30:
Looks like [pandoc package is in Ubuntu ](https://packages.ubuntu.com/focal/pandoc)so, maybe something like [this](https://code-maven.com/slides/github-ci/install-packages-on-ubuntu-linux-in-github-actions) but of course `sudo apt-get -y install pandoc` instead of tree

e.g.
```
- uses: actions/checkout@v3
#install pandoc
- name: Install pandoc package
  run: |
      sudo apt-get -y install pandoc
#old stuff
- name: autoconf
   run: autoreconf -i
```

sunwire, 2022-04-30:
@jonasstein You need to fix the workflow file, [see my file](https://github.com/sunwire/stenc/actions/runs/2250783401/workflow)
and comment on your [commit](https://github.com/scsitape/stenc/commit/41b8baec2a82986774d525a52877acc50562b394#r72571794)

jonasstein, 2022-04-30:
@sunwire DONE. Do I still miss something?

sunwire, 2022-04-30:
> @sunwire DONE. Do I still miss something?

Missing spaces before sudo apt-get

jonasstein, 2022-04-30:
Github mentions another error: in line 11
https://github.com/scsitape/stenc/actions/runs/2250941541/workflow#L11

sunwire, 2022-04-30:
@jonasstein You missed spaces before run:
> Update: But it works now. I think we can close this as solved now.
But, as I see you fixed it

closed 2022-05-01 by jonasstein