scsitape/stenc #55: New IBM LTO8 drive firmware returns 0x24 error

dalvenjah, 2022-05-03:
This looks like issue #11 but is different (this error code is 0x24 vs issue 11's 0x26).

Latest (as of May 3 2022) stenc works fine with an IBM Ultrium-TD8 drive with J4D0 and JAYE firmwares; but with the latest N4Q0 firmware, stenc returns an error:

```
$ sudo ./stenc -f /dev/nst0 --detail
Status for /dev/nst0
--------------------------------------------------
Device Mfg:              IBM
Product ID:              ULTRIUM-TD8
Product Revision:        N4Q0
Sense Code:              Illegal Request (0x05)
 ASC:                    0x24
 ASCQ:                   0x00
 Additional data:        0x00000c01504230303030340000000abb00000001080080a2704c38000000040000000000000000000000000000000000000000000000000000000000000000000000000054314d5930484756444100000000000000000000

```
Per IBM's doc at https://www.ibm.com/docs/en/ts3500-tape-library?topic=information-sense-key-illegal-request , ASC 0x24 and ASCQ 0x00 means "Invalid Field in CDB". Not sure if this means the data structure's the wrong length or what; I'll continue to do investigating on my end to see if I can work out what the drive needs with this firmware, but wanted to open this issue in case others can help take a look.

Also in case it helps, here's the link to IBM's fixlist showing what's changed between J4D0/JAYE firmwares and the latest N4Q0 firmware: https://delivery04.dhe.ibm.com/sar/CMA/STA/09qkh/0/LTO8_N4Q0_fixlist.txt

sunwire, 2022-05-05:
@dalvenjah Can you run this command and show what it returns?  `sg_raw -r 44 /dev/your_tape_dev a2 00 00 00 00 00 00 01 00 00 00`

jonasstein, 2022-07-20:
@dalvenjah ping

jonasstein, 2022-08-25:
@dalvenjah any news?

copyleftproducts, 2026-10-08:
@jonasstein on S2T1, the result is:

```
sg_raw -r 44 /dev/st0 a2 00 00 00 00 00 00 01 00 00 00
NVMe Result=0x0
Received 10 bytes of data:
 00     00 00 00 00 00 00 00 02  00 20                      .........
```