scsitape/stenc #71: Blank tape

sunwire, 2022-05-14:
I've tested the latest version (https://github.com/scsitape/stenc/commit/f5856d75915bc0e1e5cdb218b5e64267e300f652) and it  throws an error when an blank tape is in the drive.

Old behavior:
```
# stenc -f /dev/nst1 --detail
Status for /dev/nst1
--------------------------------------------------
Device Mfg:              HP      
Product ID:              Ultrium 4-SCSI  
Product Revision:        U64D
Drive Encryption:        off
Drive Output:            Not decrypting
                         Raw encrypted data not outputted
Drive Input:             Not encrypting
Key Instance Counter:    0
```
New old behavior:
```
# stenc -f /dev/nst1 --detail
Status for /dev/nst1
--------------------------------------------------
Vendor:                  HP      
Product ID:              Ultrium 4-SCSI  
Product Revision:        U64D
Drive Encryption:        off
Drive Output:            Not decrypting
                         Raw encrypted data not outputted
Drive Input:             Not encrypting
Key Instance Counter:    0
Sense Code:              Blank tape (0x08)
 ASC:                    0x14
 ASCQ:                   0x03
 Additional data:        0x00000000000000000000000000000000
```

jmwilson, 2022-05-14:
Can you try the `cleanup` branch on `jmwilson/stenc`? I'm unable to get a blank tape to trigger this sense code. The old code would ignore all errors in the next block encryption status query, so I added a media access check but my hardware will not report a blank tape.

sunwire, 2022-05-15:
> Can you try the `cleanup` branch on `jmwilson/stenc`?

Still the same error, blank tape. But it fixes https://github.com/scsitape/stenc/issues/72
If I add O_NONBLOCK flag to line https://github.com/scsitape/stenc/blob/f5856d75915bc0e1e5cdb218b5e64267e300f652/src/scsiencrypt.cpp#L92
it also fixes https://github.com/scsitape/stenc/issues/12 (under Linux)

jmwilson, 2022-05-15:
This could be a drive firmware bug then. HP docs say this about the blank tape sense key:

> If “blank media” is encountered during a space (in other words, the drive attempted to space on an unformatted tape), the drive behaves as if EOD was at BOM. It will then respond with GOOD status to a request to space to EOD, and with CHECK CONDITION to any other forward spacing request. Additional sense is set to 1403h (End of data not found).

Requesting status will not move the tape. The firmware might be incorrectly returning this code during the normal media access check when the next block encryption status page is processed. An erased tape returns `Volume Encryption:       Unable to determine` on a HP LTO-6.

The old code simply ignored all errors from the next block encryption status query, but this probably not good in general. It sounds like the best idea to ignore BLANK CHECK sense key in this specific case.

jmwilson, 2022-05-16:
@sunwire I forced pushed the cleanup branch again with a check for the BLANK CHECK sense - can you see if this handles your situation?

sunwire, 2022-05-16:
> can you see if this handles your situation?
Everything works!

sunwire, 2022-06-05:
Fixed

closed 2022-06-05 by sunwire