scsitape/stenc #104: Error when compiling for freeBSD

eccen, 2023-01-22:
I'm wondering if you might have advice on getting stenc to compile on freeBSD. I created a new jail in 13.1, installed automake and I get this error:

# autoreconf --install 
configure. ac: 15: warning: The preprocessor macro SIDC_ HEADERS' is obsolete.
configure.ac: 15: Except in unusual embedded environments, you can safely include all 
configure.ac: 15: ISO C90 headers unconditionally.
configure.ac:80: error: possibly undefined macro: AC_MSG_NOTICE
If this token and others are legitimate, please use m4 _pattern allow.
See the Autoconf documentation.
configure.ac:81: error: possibly undefined macro: AC SUBST autoreconf2.71: error: /usr/local/bin/autoconf2.71 failed with exit status: 1

jonasstein, 2023-01-23:
Did you compile the latest source from git, or a release? Which versions of the build tools and compiler do you use? Can you compile it manually?

sunwire, 2023-01-23:
@eccen did you install pkg-config?

DaveQB, 2023-03-02:
Chiming in to say I have the same error. 13.1 as well.
I have tried compiling from `master` branch (commit 0e53223fc580730485bad307519545a0c677d257).

```
automake: 1.16.5
autoconf: 2.71
autoconf-switch: 20220527
m4: 1.4.19,1

```

@sunwire 
Upon you suggestion, I tried that pkg install.

```
$ pkg install pkg-config
Updating FreeBSD repository catalogue...
FreeBSD repository is up to date.
All repositories are up to date.
pkg: No packages available to install matching 'pkg-config' have been found in the repositories
```

Thank you.

DaveQB, 2023-03-02:
Interestingly, a second run of `autogen.sh` results in a better output and a 0 exit status:

```
$ ./autogen.sh  
configure.ac:15: warning: The preprocessor macro `STDC_HEADERS' is obsolete.
configure.ac:15:   Except in unusual embedded environments, you can safely include all
configure.ac:15:   ISO C90 headers unconditionally.
configure.ac:9: installing './compile'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
src/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
```

A subsequent `./configure` results in this:

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... ./install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... nawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether cc accepts -g... yes
checking for cc option to enable C11 features... none needed
checking whether cc understands -c and -o together... yes
checking dependency style of cc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for sys/types.h... (cached) yes
checking for sys/machine.h... no
checking for g++... no
checking for c++... c++
checking whether the compiler supports GNU C++... yes
checking whether c++ accepts -g... yes
checking for c++ option to enable C++11 features... none needed
checking dependency style of c++... gcc3
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking whether to output raw SCSI messages... no
checking default encryption algorithm index to use... 0
checking your OS... FreeBSD
checking whether to build with static libgcc... no
checking for pandoc... yes
./configure: 5314: Syntax error: word unexpected (expecting ")")
```

That block of code for easy reference.

```

  5313  # BASH completion
  5314  PKG_CHECK_VAR(BASH_COMPLETION_DIR, bash-completion >= 2.0, completionsdir,
  5315  
  5316          with_bash_completion_dir=yes;
  5317          AC_MSG_NOTICE(BASH_COMPLETION_DIR is $BASH_COMPLETION_DIR);
  5318          AC_SUBST([BASH_COMPLETION_DIR])
  5319      ,
  5320      with_bash_completion_dir=no)
  5321  

```

I can't see any obvious syntax errors there.

eccen, 2023-03-02:
I eventually got it to compile by chasing dependencies. Unfortunately, I was sort of shooting in the dark, but looking at my bash history, it looks like `pkg install hs-pandoc` might have been the key.

DaveQB, 2023-03-02:
Thanks @eccen 
Yes, I came across that too actually. I forgot to mention in my message that I have installed `hs-pandoc` along the way. I think `./configure` pointed that out to me.

My issue is the syntax error in `configure`.

The next step after `autogen.sh` is `./configure` right?

sunwire, 2023-03-02:
> @sunwire
> Upon you suggestion, I tried that pkg install.

My bad package is pkgconf https://www.freshports.org/devel/pkgconf
You also need bash-completion

DaveQB, 2023-03-03:
That worked. Thanks @sunwire 

I think `autogen.sh` should detect and report these dependencies. Am I right? If so, I can look at updating `autogen.sh` and submitting a pull request.

And/or I can workout the exact packages needed and list them for the README. Would that be useful for the project?

sunwire, 2023-03-03:
> I think `autogen.sh` should detect and report these dependencies. Am I right?

Yes you are.

> I can look at updating `autogen.sh` and submitting a pull request.

Please, do.