scsitape/stenc #114: Invalid argument error when trying to use stenc in RHEL 9.3 with lin_driver (IBM TS4300)

c3rberus, 2024-02-16:
Hello,

I am using IBM TS4300 Tape Library LTO8 FC library, this is zoned to a RHEL 9.3 Linux host, the ULT3580-HH8 tape drive is recognized.

```
# rpm -qi lin_tape
Name        : lin_tape
Version     : 3.0.66
Release     : 1
Architecture: x86_64
Install Date: Thu 15 Feb 2024 02:39:41 PM
Group       : System Environment/Kernel
Size        : 1735657
License     : GPL
Signature   : (none)
Source RPM  : lin_tape-3.0.66-1.src.rpm
Build Date  : Thu 15 Feb 2024 02:38:14 PM
Build Host  : van-oppy-rhel1
Packager    : IBM Tape SCSI Device Driver Development
Vendor      : IBM
Summary     : IBM Tape SCSI Device Driver for Linux
Description :

# rpm -qi lin_taped
Name        : lin_taped
Version     : 3.0.66
Release     : 1
Architecture: x86_64
Install Date: Thu 15 Feb 2024 02:40:59 PM
Group       : System Environment/Kernel
Size        : 179136
License     : Proprietary
Signature   : (none)
Source RPM  : lin_taped-3.0.66-1.src.rpm
Build Date  : Mon 28 Aug 2023 09:29:31 AM
Build Host  : redondo.tuc.stglabs.ibm.com
Packager    : IBM Tape SCSI Device Driver Development
Vendor      : IBM
Summary     : IBM Tape SCSI Device Driver Application/Daemon
```

I have installed the IBM Tape SCSI Device Driver for Linux (lin_tape), and I can write/read to the device using tar.

However, when I try to use stenc to check status, I get an error.

```
# stenc --v
stenc 2.0.0 - SCSI Tape Encryption Manager

# stenc -f /dev/IBMtape0
stenc: Invalid argument

**# dmesg**
[14164.410790] lin_tape: lin_tape_drive_ioctl invalid ioctl
```

Anyone have any idea on what could be going on?

c3rberus, 2024-02-16:
Just to add to this, debug lin_tape driver logs below.

```
lin_tape: lin_tape_open entry: minor 3071
lin_tape: lin_taped is already opened.
lin_tape_open exit: -11
lin_tape: lin_tape_open entry: minor 0
lin_tape_open: IBMtape0 using current path 0
lin_tape: tape_inquiry_page entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x12
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_tur entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x0
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0xd2
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape_write_dynamic_attrs: 0
lin_tape: tape_reserve entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x16
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_get_media_parm entry
lin_tape: tape_modesense_page entry: 0x10
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x1a
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_modesense_page exit: 0
lin_tape: tape_get_media_parm exit: 0
lin_tape: tape_update_block_limits entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x5
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_get_encryption_capability entry.
lin_tape: tape_modesense10_page entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x5a
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_modesense10_page exit: 0
lin_tape: tape_get_encryption_capability exit: 0
IBMtape0-----0805B: encryption capable is 1
lin_tape: tape_get_encryption_enablement entry.
lin_tape: tape_modesense10_page entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x5a
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_modesense10_page exit: 0
IBMtape0-----0805B: encryption enablement is 1
lin_tape: tape_encryption_write_setting entry
lin_tape: tape_encryption_write_setting exit: 0
lin_tape: tape_get_pos entry: fmt 0 ex 0
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x34
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: current position 0
lin_tape: tape_get_pos exit: 0
lin_tape: amu entry
lin_tape: tape_execute_scsi_command entry: 0x5a
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: amu exit: 0 amu is 0
lin_tape: lin_tape_drive_open exit: 0
lin_tape_open exit: 0
lin_tape_ioctl entry: minor 0 command 0x2285
lin_tape: Drive busy: 0 flag 1
lin_tape: lin_tape_drive_ioctl invalid ioctl.
lin_tape_ioctl exit: -22
lin_tape_close entry: minor 0
lin_tape: lin_tape_drive_close entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0xd2
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape_write_dynamic_attrs: 0
lin_tape: tape_rewind entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x1
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: tape_release entry
lin_tape: tape_send_scsi_cmd entry.
lin_tape: tape_send_scsi_io entry
lin_tape: tape_execute_scsi_command entry: 0x17
lin_tape: tape_execute_scsi_command exit: 0x0
lin_tape: tape_check_result entry.
lin_tape: tape_check_result exit: 0
lin_tape: tape_send_scsi_io exit: 0
lin_tape: tape_send_scsi_cmd exit: 0
lin_tape: lin_tape_drive_close exit: 0
lin_tape_close exit: 0
```

c3rberus, 2024-02-16:
If I remove lin_tape and go back to standard linux SCSI generic (sg) driver and SCSI tape (st) driver, seems to work.

```
# ./stenc -f /dev/st0
Status for /dev/st0 (IBM ULT3580-HH8 M571)
--------------------------------------------------
Reading:                         Decrypting (AES-256-GCM-128)
                                 Unencrypted blocks not readable
Writing:                         Encrypting (AES-256-GCM-128)
Key instance counter:            1
Drive key desc. (U-KAD):         XXXXXX
Current block status:            Encrypted and able to decrypt (AES-256-GCM-128)
Supported algorithms:
1    AES-256-GCM-128
     Key descriptors allowed, maximum 32 bytes
     Raw decryption mode allowed, raw read enabled by default
2    AES-256-GCM-128
     Key descriptors allowed, maximum 32 bytes
     Raw decryption mode allowed, raw read enabled by default
3    AES-256-GCM-128
     Key descriptors allowed, maximum 32 bytes
     Raw decryption mode allowed, raw read enabled by default
```

I am guessing lin_tape driver is not supported?

rogerbrowm, 2025-02-23:
I have the same problem:
 tape_dev.c:368-3183 tape_dev.c:366 ioctl MTIOCGET error on "Drive-1" (/dev/IBMtape1n). ERR=No such file or directory.
mount.c:318-3183 Unable to position to end of data on Tape device "Drive-1" (/dev/IBMtape1n): ERR=tape_dev.c:366 ioctl MTIOCGET error on "Drive-1" (/dev/IBMtape1n). ERR=No such file or directory.
Is there a solution for that ?

c3rberus, 2025-02-28:
I never found a solution to this other than not using lin_driver for tape.

amihaisilver, 2025-06-30:
This is correct. 
Only after the Lin-tape driver is removed from the operating system, the mt-st devices are seen (in by the sg kernel module).
It seems as lin_tape overrides the st devices. The IBM LTO9 tape devices works properly with bareos by the st devices. 
The IBM tape drives user manual instruct to install lin_tape IBM drivers. "Do not read the instructions" :-)))